matt1795/BCIT-Sailbot #3: Decide between Mono-hull and Catamaran

matt1795, 2016-10-06:
Reasons for choosing Catamaran:
- Decrease surface drag
- Increase Stability
- Increase Maneuverability

Reasons for choosing Mono-Hull:
- Simpler construction

I am in favor of a Catamaran style design. Manufacturing is slightly more difficult but the  performance would definitely be worth it. I'd like to hear what you guys have to say, and some more perceived  pros and cons for either hull type.

James-Thegiant, 2016-10-06:
Definitely in favour of the catamaran style. I also think Craig's suggestion of hydrofoil design is worth looking into.

matt1795, 2016-10-06:
I'll make a ticket for the Hydrofoil.

matt1795, 2016-10-09:
http://www.harborwingtech.com/HWT-X-3-Production-Design.htm

This also intrigues me. Both the hull, sail, and hydrofoil design look fantastic.

James-Thegiant, 2016-10-10:
I had been going to mention that in the (admittedly not a huge amount of) research I've done so far, the fastest sailboats are consistently trimarans

matt1795, 2016-10-12:
We will be going with a monohull

closed 2016-10-12 by matt1795