matt1795/BCIT-Sailbot #5: Find a Wind Sensor

matt1795, 2016-10-06:


Xthreeo, 2016-10-07:
**Viktor**
$17.00
https://moderndevice.com/product/wind-sensor/

Logarithmic detection via heating of a component to a constant temperature and then measuring the electrical power that is required to maintain the heated element at temperature as the wind changes. This measured electrical input is then directly proportional to the square of the wind speed.

Claim up to 60 mph detection with high precision

matt1795, 2016-10-07:
Unfortunately the hot wire anemometer would only measure wind velocity, not direction. One of the more popular solutions that I'm seeing sailboats use are Ultrasonic wind sensors:

http://hackaday.com/2013/08/21/ultrasonic-anemometer-for-an-absurdly-accurate-weather-station/#comments

This is DIY, but we might be able to make this into a project for some of the Robotics guys as a term project. Would also cut down on costs quite a bit.

Xthreeo, 2016-10-07:
Nice find! Ya looks like an awesome project for Robotics guys

cameron120, 2016-10-08:
Looking at online nautical stores (ie. http://www.defender.com/category.jsp?path=-1%7C17%7C2028690&id=2028831), a common design seems to be the wind vane/anemometer combo. 

We could make a wind vane and attach it to a rotary encoder to determine direction. Ditto for the anemometer.

closed 2017-04-26 by matt1795