matt1795/BCIT-Sailbot #6: Create Testing Apparatus for Vision System

matt1795, 2016-10-06:
In order to test an implement a 2-camera vision system that will have depth perception, we will need:

- 2x4 - approximately 1 meter across
- Raspberry Pi
- Two webcams
- 9 axis IMU

Xthreeo, 2016-10-07:
**Viktor**

For Cameras 
**CAD $90.23 ea**

1280x800 Video Resolution
http://www.robotshop.com/ca/en/pixy-cmucam5-image-sensor.html

On-board image processing.
***Colour filtering built-in***
http://www.robotshop.com/media/files/images2/pixy-cmucam5-image-sensor-10.gif

lowkin18, 2016-10-13:
are there color requirements for the boat?
An orange boat would add complexity and something we would have to plan for.

I think we are looking for something like the Jetson Tk1($240) to power the vision system if we want collision avoidance and potentially the Tx1($600).

I have a Tk1 that I will be playing around with and i'll see the exact performance. On the specs sheet it says it can handle 32FPS at 720P but I dono.

I have very limited knowledge on OPENCV, I can track an object by color or shape or the template haarcascades.

terrycalderbank, 2016-10-18:
**Terry**
I've ordered a 9 axis IMU to mess with, will let you guys know when it arrives.

matt1795, 2017-04-14:
Will discuss this in the next meeting

closed 2017-04-14 by matt1795