matt1795/BCIT-Sailbot #8: Sourcing a GPS Module

matt1795, 2016-10-06:
We need to figure out how accurate of a GPS Sensor we can get, and what sort of cost are we going to pay for that accuracy?

Xthreeo, 2016-10-06:
**$30 Canadian dollars off Amazon**

If cheaper than a more precise option, we can get multiple, and compute/compare location off more than one.

https://www.amazon.ca/High-precision-High-sensitivity-Digital-Electronic-Navigation/dp/B01K78ESS4/ref=sr_1_13?ie=UTF8&qid=1475785781&sr=8-13&keywords=high%2Bprecision%2Bgps&th=1

**Datasheet** 
http://www.elechouse.com/elechouse/images/product/GY-26-USART%20Digital%20Compass/gy-26%20manual.pdf

matt1795, 2016-10-07:
I really like the idea of multiple GPS sensors and averaging them. However the link you gave is for a compass sensor which will only gives a heading. I'd like to use a cheap 9 axis IMU for that since we also get an accelerometer and gyroscope.

http://www.robotshop.com/ca/en/bno055-9-dof-absolute-orientation-imu-fusion-breakout-board.html

matt1795, 2017-04-14:
GPS sensor has been sourced.

closed 2017-04-14 by matt1795